SonnenladenGmbH/APsystems-EZ1-API #2: Probleme mit pip install apsystems-ez1

walliesp, 2023-11-28:
Mit meinem Linux Mint Rechner VERSION="19.3 (Tricia)"und mit Python 3.6 war ich in der Lage  mit dem Befehl

pip install apsystems-ez1

die entsperchenden Dateien runterzuladen und auch ein Python 3.6 Programm zum laufen zu bringen, das die Wechselrichterdaten ausliest.
Nun habe ich es nicht geschafft  mit  einer  höheren Python (3.10) Version das gleiche Programm zum Laufen zu bringen, da  damit die  die ensprechen Dateien von Github nicht gefunden werden. Momentan sind diese in den Verzeichnisssen 

.local/lib/python3.6/site-packages/apsystems_ez1-0.1.1.dist-info
und in /.local/lib/python3.6/site-packages/APsystems_EZ1

mit python 3.10 sieht die Ausgabe dann folgendermassen aus:

******
$python3.10 /home/ryzen-mint-ssd/Python_Programme/wechselrichter_EZ1M/main.py
Traceback (most recent call last):
  File "/home/ryzen-mint-ssd/Python_Programme/wechselrichter_EZ1M/main.py", line 1, in <module>
    from APsystems_EZ1 import APsystemsEZ1M
ModuleNotFoundError: No module named 'APsystems_EZ1'
******

während mit python3.6 daten vom Wechselrichter kommen:

******
$ python3.6 /home/ryzen-mint-ssd/Python_Programme/wechselrichter_EZ1M/main.py
deviceinfo: {'data': {'deviceId': 'E07000078790', 'devVer': 'EZ1 1.6.1', 'ssid': '*****', 'ipAddr': '****', 'minPower': '30', 'maxPower': '800'}, 'message': 'SUCCESS', 'deviceId': '*****'}
gesamteenergie: 22.151130000000002
get_output_data(): {'data': {'p1': 11, 'e1': 0.04313, 'te1': 11.2066, 'p2': 11, 'e2': 0.04065, 'te2': 10.94453}, 'message': 'SUCCESS', 'deviceId': '*****'}
<class 'dict'>
3
{'p1': 11, 'e1': 0.04313, 'te1': 11.2066, 'p2': 11, 'e2': 0.04065, 'te2': 10.94453}
{'p1': 11, 'e1': 0.04313, 'te1': 11.2066, 'p2': 11, 'e2': 0.04065, 'te2': 10.94453}
<class 'dict'>
6
p1:  11 W
p2:  11 W
e1:  0.04313 kWh
e2:  0.04065 kWh
******

jetzt könnte ich mich ja damit zufrieden geben, aber ich will natürlich nicht den Desktop-Rechner Tag und Nacht laufen lassen.

Also habe ich ein Raspberry 4 genommen und die Neueste Version frisch aufgespielt

PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"

Python 11 läuft:

******
$ python -V
Python 3.11.2
******

PIP auch:

******
$ pip -V
pip 23.0.1 from /usr/lib/python3/dist-packages/pip (python 3.11)
******

aber dann komme ich nicht weiter:

PIP kann die Bibliotheken nicht herunterladen:

******
$ pip install apsystems-ez1
error: externally-managed-environment

This environment is externally managed

To install Python packages system-wide, try apt install
python3-xyz, where xyz is the package you are trying to
install.

If you wish to install a non-Debian-packaged Python package,
create a virtual environment using python3 -m venv path/to/venv.
Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
sure you have python3-full installed.

For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
*****

Vielleich gibt es andere, die ähnliche Probleme haben

mawoka-myblock, 2023-11-28:
Du musst erst [Pipx](https://pypa.github.io/pipx/) installieren.
`sudo apt install python3-pipx` und dann kannst du die Library mit `pipx install apsystems-ez1`

Des Weiteren ist dies kein Problem mit dieser Bibliothek. Falls du dennoch weitere Probleme haben solltest, frag einfach nach.

closed 2023-11-28 by mawoka-myblock

walliesp, 2023-11-28:
$ sudo apt install python3-pipx
Paketlisten werden gelesen... Fertig
Abh�ngigkeitsbaum wird aufgebaut... Fertig
Statusinformationen werden eingelesen... Fertig
E: Paket python3-pipx kann nicht gefunden werden.

geht leider auch nicht

CM000n, 2023-11-29:
Laut requirements in der `pyproject.toml` wird Python >= 3.11 vorausgesetzt.

walliesp, 2023-11-29:
Also bei mir läuft das auf Python 3.6 auf meinen Linux Mint 19.3
und jetzt auch auf meinem Raspberry zero  "Raspbian GNU/Linux 11 (bullseye)"
unter Python 3.9.2
### **ABER DAS IST auch nicht mein PROBLEM:**
### **Das Problem ist dass die pip installation nur einmal funktionierte** auf dem Desktop  Linux Mint 19.3 und dann nicht mehr wie oben beschrieben
Ich habe jetzt einfach die Verzeichniss  mit einem USB-Stick von dem linux Mint auf den raspberry zero händisch kopiert.
Da habe ich ganze drei Tage dran gesessen bis ich herausgefunden habe dass es so geht.
### **Das Problem ist dass pip mit den neuen Raspberryversionen(die dann auch die höheren Python Versionen beinhalten) nicht zu funktionieren scheint.**
Siehe hier :https://www.raspberrypi.com/documentation/computers/os.html#using-pip-with-virtual-environments
### **ZITAT**
"_About Python virtual environments

### **In previous versions of the operating system, it was possible to install libraries directly, system-wide, using the package installer for Python, commonly known as pip.** You’ll find the following sort of command in many tutorials online.

$ pip install xxxxx

**In newer versions of Raspberry Pi OS, and other operating systems, this is disallowed**. If you try and install a Python package system-wide you’ll receive an error similar to this:

$ pip install buildhat
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
  python3-xyz, where xyz is the package you are trying to
  install.

  If you wish to install a non-Debian-packaged Python package,
  create a virtual environment using python3 -m venv path/to/venv.
  Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
  sure you have python3-full installed.

  For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

This error is generated because you’re trying to install a third-party package into the system Python. A long-standing practical problem for Python users has been conflicts between OS package managers like apt and Python-specific package management tools like pip. These conflicts include both Python-level API incompatibilities and conflicts over file ownership.

Therefore from Bookworm onwards, packages installed via pip must be installed into a Python virtual environment using venv. A virtual environment is a container where you can safely install third-party modules so they won’t interfere with, or break, your system Python."_
### **ZITAT-ENDE**
Ich habe das gemäss Anleitung über
virtual environments
versucht was aber leider nicht funktioniert hat