SonnenladenGmbH/APsystems-EZ1-API #8: Feature Request: Input voltage and power limit per input and prio Input

MhlTheOne, 2024-01-07:
First of all, thank you for the locale API.

To connect this inverter to the battery on one input and directly to the panels on the second, the following features would be desirable:

1. Input voltage for each Input (to save Battery) or set min voltage per Input
2. Powerlimit for each Input (to set Solar Input max and Battery Input to need for zero supply)
3. Or something like an Prio Input with total limit (so first use solar and for difference use Battery)

mawoka-myblock, 2024-01-07:
Since we aren't associated with APsystems, @SonnenladenGmbH could at most ask them, but that's all we can do, so I'll close this issue.

MhlTheOne, 2024-01-07:
Yes you aren't APsystems, but you are in contact and promote the API.

SonnenladenGmbH, 2024-01-07:
Thanks for the request @MhlTheOne 

We developed the API in close partnership with APsystems and will forward your feature request to them asap. 

This is a place to request features for the API just like you did it.

Best
Leo

mawoka-myblock, 2024-01-07:
Then apologies from my side for turning you off ro o quickly and harsh! @MhlTheOne

ThomasRgbg, 2024-02-26:
Getting the input voltage would be a really nice feature. I hope its not a lot of extra work for APSystems. (And getting a fixed target voltage would then maybe already to much to dream... :)

gh4tg, 2025-03-17:
Okay, one year later, as far as I see there is no progress in adding new datapoints like the input voltage?
Thanks, kind regards, thg

mawoka-myblock, 2025-03-17:
Feel free to reach out to APsystems on your own, there's nothing else I can do here.