SonnenladenGmbH/APsystems-EZ1-API #9: Request ability to enable via API and subsequently via Homeassistant

thowiegit, 2024-01-26:
Es wäre schön wenn man Bluetooth über die lokale API bei Bedarf wieder einschalten kann.

Aktuell muss man kurzzeitig den Strom zum Inverter trennen oder sein Wifi vorübergehend abschalten um Bluetooth bei Bedarf wieder zu aktivieren, das ist doch eher umständlich.

Es wäre daher schön dies mit über die lokale API und aus Homeassistant heraus erledigen zu können.

Denn schließlich wird man Bluetooth nicht dauerhaft aktiv lassen wollen um dritten keine Spielwiese offen zu lassen.

DirkBaumeister, 2024-04-05:
Just adding my comment here also to support this request :) Please let us know if there is any progress :)

Schw3n, 2024-06-06:
Dear all, 
I would like to take this opportunity to briefly describe my experiences. I was very pleased to convince my neighbor to buy a plug-in solar device.
Of course, he also bought an AP System EZ1. As luck would have it, I've just received a text message from him saying “hey, I've just dialed into your inverter via bluetooth.”
If my neighbor can do it, what are strangers capable of?

I would therefore like to support this request.

amjadsaadeh, 2024-09-26:
Yes having the ability to turn off bluetooth is quite important. But it looks like that the API provided by EZSystems does not support this, at least couldn't find any docs about this.

lukas-fichtner, 2025-05-20:
Any known updates from the vendor about this feature?