SonnenladenGmbH/APsystems-EZ1-API #11: MicroPython port

tacker66, 2024-02-09:
I am trying to call the local API from MicroPython using only the `requests` module.
[My approach](https://github.com/tacker66/pihome/blob/master/MicroPython/Power/ez1.py) works with CPython but needs a patched `requests ` module for MicroPython. I raised an [issue ](https://github.com/micropython/micropython-lib/issues/802) in micropython-lib describing the problem. Perhaps you could help here?

mawoka-myblock, 2024-02-09:
I'm not a huge micropython fan and it's really annoying to get all the files on the esp. This also doesn't have much to do with this library. Wouldn't it just be possible to replace [these](https://github.com/SonnenladenGmbH/APsystems-EZ1-API/blob/f56330d76d831277f9178a8becd1b6db7228b459/APsystemsEZ1/__init__.py#L69-L73) lines with another request-client?

tacker66, 2024-02-09:
I created this issue because you stated that you are working on a MP port ;-) 
Problem is there is no `urllib3 `or `aiohttp` for MP and `requests` consist only of a minimal implementation for MP.
BTW: With the new `mip `installer bringing files to an ESP32/PicoW is quite easy (see my link above)..

tacker66, 2024-02-09:
Yeah, but the problem seems to lie in the underlying MP socket implementation (at least my guess).