SonnenladenGmbH/APsystems-EZ1-API #13: Feature Request: Allow for dynamic output power adjustment with fallback setting (allows zero-feed-in or limited feed-in)

NeoFromMatrix, 2024-02-23:
It would be nice if the API would enable users to build a custom setup which limits the grid feed-in. 
This could be done via dynamically setting the output power in combination with a persistent fallback power setting.
A separate device would be needed to measure power across the electricity meter and to calculate the currently allowed power setting for the inverter(s). This is not part of the feature request. 

The fallback power setting should trigger if no dynamic updates are received for a given timeout. 
This ensures a lost network connection or connection to the controlling device do not result in unpermitted grid feed-in.

Further, a method for setting the max. power of the inverter is needed which reacts fast enough and does not wear down any potential internal memory chips.

mawoka-myblock, 2024-02-23:
Sounds reasonable, but I think that APsystems doesn't think it would be worth the effort.

TipsTricksMore, 2024-02-23:
We talked to APsystems last week.
They will enable such feature in an upcoming firmware and app release.

NeoFromMatrix, 2024-02-23:
Oh, very nice to hear!
It could actually be implemented with a single new API call. 

The existing set_max_power() could be used to set the persistent "fallback" power to a low value (e.g. 30W). 

A new set_temporary_max_power() API call could be used to set the dynamically higher power. 
This could include a fixed timeout or a second parameter which specifies the timout. 
Within the given timout it should be possible to overwrite/update the max_power with the same API call which would also reset the timer. 
Again, if the timeout is reached, the device is rebooted or the wireless communication is lost, the device would fall back to the previous defined low limit.

Software complexity, requirements and integration wise this should come with a very small footprint.

ccvca, 2024-06-20:
> We talked to APsystems last week. They will enable such feature in an upcoming firmware and app release.

@TipsTricksMore  Any news on this?
Is there any public available change log of firmwares?