SonnenladenGmbH/APsystems-EZ1-API #14: Debug Output in "Get Device Info"

Sebastian-MrBeam, 2024-03-09:
### What happened?

response = await self._request("getDeviceInfo")
        _**print(response)**_
        return (
            ReturnDeviceInfo(
                deviceId=response["data"]["deviceId"],
                devVer=response["data"]["devVer"],
                ssid=response["data"]["ssid"],
                ipAddr=response["data"]["ipAddr"],
                minPower=int(response["data"]["minPower"]),
                maxPower=int(response["data"]["maxPower"]),
            )
            if response and response.get("data")
            else None
        )

### The python error

```shell
Every function call prints an Output - all other functions does not, so looks like a debug leftover.
```

### Your python version

3.12

### Your OS inlcuding your version

Windows 11

mawoka-myblock, 2024-03-13:
Indeed. Shouldn't be there, thank you for noticing that!

mawoka-myblock, 2024-04-04:
I've removed that in the version 3 release.