SonnenladenGmbH/APsystems-EZ1-API #16: [Bug]:  Cannot connect to host 192.168.1.212:8050 ssl:default [The remote computer refused the network connection]

paazevedo, 2024-03-29:
### What happened?

While trying to execute the following example, I get an SSL issue:

from APsystemsEZ1 import APsystemsEZ1M # import the APsystemsEZ1 library
import asyncio

# Initialize the inverter with the specified IP address and port number.
inverter = APsystemsEZ1M("192.168.1.212", 8050)

async def main():
    try:
        # Get device information
        device_info = await inverter.get_device_info()
        print("Device Information:", device_info)

        # Get alarm information
        alarm_info = await inverter.get_alarm_info()
        print("Alarm Information:", alarm_info)

        # Get output data
        output_data = await inverter.get_output_data()
        print("Output Data:", output_data)

        # Set maximum power limit (ensure the value is within valid range)
        set_power_response = await inverter.set_max_power(500)
        print("Set Power Response:", set_power_response)

        # Set power status (ensure "ON" or other value is valid)
        set_power_status_response = await inverter.set_device_power_status("ON")
        print("Set Power Status Response:", set_power_status_response)

        # Get current power status
        power_status = await inverter.get_device_power_status()
        print("Power Status:", power_status)
    except Exception as e:
        print(f"An error occurred: {e}")

# Run the main coroutine
asyncio.run(main())

### The python error

```shell
PS C:\Temp> python.exe .\EZ1.py 
An error occurred: Cannot connect to host 192.168.1.212:8050 ssl:default [The remote computer refused the network connection]
PS C:\Temp>
```

### Your python version

3.11.8

### Your OS inlcuding your version

Windows11

mawoka-myblock, 2024-03-29:
Are you sure the IP address is correct? This error seems to occur when the IP is wrong.

paazevedo, 2024-03-29:
Looks like. Unfortunately from the APSystems application I can't determine which IP\Mac address the device has, but from my unifi management I see it's apsystems. Used other techniques as well, as reconnect the device and see which devices joins the network for the less time. So I'm sure the IP address. 

[image]

mawoka-myblock, 2024-03-29:
What happens if you open http://192.168.1.212:8050 ? It should give you an empty page saying "Nothing matches the given URI"

paazevedo, 2024-03-29:
> What happens if you open http://192.168.1.212:8050 ? It should give you an empty page saying "Nothing matches the given URI"

I'll give a try tomorrow and report back. I don't have more sunlight, the inverter just shutdown some minutes ago.

mawoka-myblock, 2024-03-29:
Ah, I know that! Tomorrow will be the last day I'll be able to help you until next weekend.

paazevedo, 2024-03-29:
> Ah, I know that! Tomorrow will be the last day I'll be able to help you until next weekend.

That's OK, I appreciate your input. Would you recommend other method to double check the IP address of the inverter ? It returned to life again (it's cloudy and rainning), but when I do http://192.168.1.212:8050/ the connection seems to be simply refused with ERR_CONNECTION_REFUSED

mawoka-myblock, 2024-03-29:
Is the API activated in the app?

paazevedo, 2024-03-29:
> Is the API activated in the app?

How do I check that sorry.

paazevedo, 2024-03-29:
> Can't tell you right now, but on the HA-integration docs, there's the German tutorial which should help you: https://github.com/SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant
> 
> For that, you'll have to connect via Bluetooth to the inverter and enable it somewhere in the settings.

Appreciate your support! That was the problem. It's working now!

[image]

mawoka-myblock, 2024-03-29:
No problem!

closed 2024-03-29 by mawoka-myblock