SonnenladenGmbH/APsystems-EZ1-API #17: Feature request: Disable bluetooth by local api

DirkBaumeister, 2024-04-03:
As far as I have read online (Don't own this device yet) the inverter has an open bluetooth for direct connection while in local mode. There seems to be an option in cloud mode to disable bluetooth.

It would be amazing to have an api feature to disable bluetooth when in local mode.

Actually thats the only bummer for me right now that's stopping me from buying an EZ1. I'm currently choosing between hoymiles and APSystems and have a slight favor for the EZ1 because I don't like the CLEAR TEXT DTU transmission of the hoymiles inverters.
In my understanding I can set the EZ1 in a closed VLAN and just use it locally by API.

So an always open bluetooth is for me like the unencrypted DTU traffic of hoymiles (if not worse).

**So: Is it possible to disable bluetooth in local mode or any chance for this feature?**

mawoka-myblock, 2024-04-03:
@SonnenladenGmbH, let your contacts play ;)

@DirkBaumeister This could take some time, maybe a month or even longer

DirkBaumeister, 2024-04-03:
I'm not in a rush on this one :) But would be amazing actually if this could be implemented!

mawoka-myblock, 2024-04-03:
I mean, I'd love to implement that, but that's not on my end, unfortunately...

closed 2024-04-04 by mawoka-myblock