SonnenladenGmbH/APsystems-EZ1-API #20: New ClientSession handling in V1.2

Sebastian-MrBeam, 2024-04-06:
The new _request() Funktion is not closing the ClientSession  (no matter if external or self generated). 
The "with"  statement should do that automatically - in my opinion -  but somehow does not. Finally it´s necessary behavior,  because in difference to V1.1 were every  _request() call was creating a new ClientSession() in V1.2 this is done only once at class \_\_init\_\_ . If a ClientSession can be closed by errors  the EZ1M class (without external Session handling ) will stop working - monitoring  will show if this is a relevant issue.

Overall the calls Monitoring for last 2 days shows less missed calls than V1.1 (on a daily base) - more stable = nice!

One idea for sanitizing  - if a Funktion opens a  Instanz of EZ1M (without external Session handling ) a open Session remains  and raises an error when Funktion ends. This can be manually fixed with " EZ1M.session.close()  
A class Funktion  could be a improvement - EZ1M.close()  is still manual work

```
async def close(self):
    if self.session.closed == False :
           await self.session.close()
```

the  class  __del__  call  looks not compatible to async, maybe someone with more experience have a idea to automate that for a careless destruction.

mawoka-myblock, 2024-04-06:
Fixed that in #19

closed 2024-04-06 by mawoka-myblock

Sebastian-MrBeam, 2024-04-07:
Nice solution! Thanks  for fixing 😃