SonnenladenGmbH/APsystems-EZ1-API #21: [Bug]: Client Session not closed on timeout

ccMatrix, 2024-04-08:
### What happened?

Over night, when the inverter is offline the requesta are running into timeout. This leads to an error in Home Assistant:
```
Logger: homeassistant
Source: runner.py:146
First occurred: 21:55:29 (39 occurrences)
Last logged: 22:00:19

Error doing job: Unclosed client session
```

There probably needs to be a try/except call around the actual request to capture timeouts and close the session.

### The python error

```shell
File "/usr/local/lib/python3.12/site-packages/APsystemsEZ1/__init__.py", line 229, in get_max_power
    response = await self._request("getMaxPower")
  File "/usr/local/lib/python3.12/site-packages/APsystemsEZ1/__init__.py", line 76, in _request
    ses = ClientSession()
```

### Your python version

3.12

### Your OS inlcuding your version

Ubuntu 22.04 Home Assistant 2004.4.1 in docker

mawoka-myblock, 2024-04-08:
HA 2004.4.1 sounds interesting ;)

Yea, I'm aware of the issue and will address it soon.

mawoka-myblock, 2024-04-09:
Fixed in https://github.com/SonnenladenGmbH/APsystems-EZ1-API/releases/tag/1.3.1

closed 2024-04-09 by mawoka-myblock