SonnenladenGmbH/APsystems-EZ1-API #24: EZ1-H

lostOzone, 2024-04-21:
It works but can't use it to it's full potential. It needs to extend power to 960

Clement-B, 2024-05-29:
Any plan to update this library and make it compatible with H model ? 

It will need to be able to identify model and I dont know if it's currently possible.
If so, we could then define max power accordingly ( M => 800, H => 960).

mawoka-myblock, 2024-08-03:
I want that to work as well. For that, I need your help: Please post the response of `http://YOUR_INVERTER_IP:8050/getDeviceInfo` here, as I personally only have the EZ1-M to develop.

lostOzone, 2024-08-05:
> I want that to work as well. For that, I need your help: Please post the response of `http://YOUR_INVERTER_IP:8050/getDeviceInfo` here, as I personally only have the EZ1-M to develop.

{"data":{"deviceId":"E05000012345","devVer":"EZ1 1.7.3","ssid":"Livebox-Zen","ipAddr":"192.168.1.101","minPower":"30","maxPower":"960"},"message":"SUCCESS","deviceId":"E05000012345"}

lostOzone, 2024-08-05:
I can change it with this URL
http://192.168.1.101:8050/setMaxPower?p=960

lostOzone, 2024-08-05:
On my second EZ1-H I get this:
{"data":{"deviceId":"E05000123456","devVer":"","ssid":"Livebox-Zen","ipAddr":"192.168.1.107","minPower":"30","maxPower":"960"},"message":"SUCCESS","deviceId":"E05000123456"}

closed 2024-08-14 by mawoka-myblock