SonnenladenGmbH/APsystems-EZ1-API #25: Local Mode not available

Kitekater, 2024-06-14:
Hi everyone,

as done in the manual on this site, i want to activate the Local mode, but the button is not shown in my app.
Whats wrong?
I installed the PV today and want to integrate it in my Home Assistant

Greetings Marc

TheKarpati, 2024-06-16:
Firmware needs to be 1.7.1 at least

dkal1, 2024-07-05:
Hi,
I have the same problem. Modellname: EZ1 Version: EZ1 1.7.5
Any ideas what I am doing wrong?
Regards Dirk

TheKarpati, 2024-07-05:
You need to completely log out of the account in your app and then make a local connection

kaiwasaki, 2024-11-26:
I am logged out and can connect to the device but have no "local mode" button.

marcussiegl, 2024-12-15:
I have completely logged out if my account and reconnected to my EZ1-M via (Direkte Verbindung). 
Problem is, hat the App of APsystems thinks that my firmwareversion of 1.2.0_B (Zendure) would be the latest version. There is still no Local-Mode and any port-scans are unsuccessfull. 

As far as i see the Wifi-controller is an espressif Microcontroller which could be flashed easyly if there would be an serial or usb-port, but at first sight the EZ1-M is solid encased in epoxy.

Maybe i will take out the big boy tools and try to wireshark my way in.

mawoka-myblock, 2024-12-15:
> I have completely logged out if my account and reconnected to my EZ1-M via (Direkte Verbindung). Problem is, hat the App of APsystems thinks that my firmwareversion of 1.2.0_B (Zendure) would be the [...]

if you choose the brutal method, please post some pictures ;) But maybe create a discussion for that part.

kaiwasaki, 2024-12-22:
I emailed APsystems to inquire about firmware update options for my device, as I couldn't find a manual update feature either. They replied that they would be rolling out updates themselves next year, (possibly through a remote process).

I followed up to clarify this unusual approach, as I believe allowing users to update their own firmware is more standard practice. Unfortunately, I haven't received a response to my query.

I'm quite disappointed with this purchase, as the availability of an API was a major factor in my decision to buy this device.

jedie, 2024-12-22:
It would be great, if somebody will hack into this and checked how the app activates the local mode. Maybe its possible to do the same with a simple Python script.

pingufreak, 2025-02-02:
Hmm have the same problem and I'm a developer / embedded engineer. Would be interesting to hack it. Any news on the latest firmware? Will they do something?

mawoka-myblock, 2025-02-02:
My tip: Just reach out to the customer service. I've heard lots positive stuff about them quickly enabling the local mode or updating the firmware or something.

pingufreak, 2025-02-02:
I've written an e-mail to the APSystems support. I will let you know, if they have enabled the update for me.

kaiwasaki, 2025-02-02:
They simply wrote back to me that they update remotely, early 2025 (I wrote that mail in November). I am still pissed.

That's the reply I got:
> Unsere Forschungs- und Entwicklungsabteilung führt Firmwareupdates per Fernwarting in Ihrer Region durch. Bitte  melden Sie sich Begin 2025, wenn das Problem noch besteht. Vielen Dank.

Which translates to:
"Our research and development department is conducting remote firmware updates in your region. Please contact us at the beginning of 2025 if the issue still persists. Thank you very much."

I don't even like the idea of a remote update (or even waiting patiently), what if the system loses power during that time?

mawoka-myblock, 2025-02-02:
Yea, the intransparency is not cool. But regarding your concern about power loss: The built-in ESP32 can have two partitions for firmware updates: One for live and one for the update. If the update the doesn't call the "I am good, the update worked" function, it automatically falls back to the previous version (and partition). This doesn't have to be like that but I am pretty sure they handle it like that as they are probably aware of the risk and all of their other programming seems quite solid.

Regarding the intransparency: I received a beta firmware for the local API one week before the public release: "Give us your serial number, tomorrow is the new firmware on your inverter". That how it worked, though I've never been in contact with them directly.

pingufreak, 2025-02-03:
Yesterday I've asked the support and gave them the serial nummer of the inverter. Today I've received the update and installed the integration in home-assistant. It works. Thank you APSystems for the great support.

pingufreak, 2025-02-03:
Version 2.0.1_b

marcussiegl, 2025-02-03:
@pingufreak do you use the APSystems App to connect to the EZ1-M ? First i tried the APSystems app and that worked, but i was not able to update the Firmware (already newest Firmware 1.2.0b installed). Then I configured the zendure app but now i can only use the APSystems App in direct connect mode (Bluetooth). Do i have to remove the inverter from the Zendure app?

pingufreak, 2025-02-03:
Please ask the support and give them your serial number of the inverter. They will than release it individually for you. Your version ist he old one, which is  1.2.0_b, I’ve received an update to 2.2.0_b.

SonnenladenGmbH, 2025-02-03:
The inverter will switch to the "_b" firmware automatically after a battery is connected. Even when it is a DIY battery.
We talked to APsystems some time ago about that...they had justified concerns because their warranty claims were not designed with the inverter running on 100% 24/7 in mind.

After a lot of talking we managed to push them to release a new "_b" firmware that supports the API now.
They're currently not rolling out the firmware automatically but will do so very soon.

At the moment every customer wanting the new firmware early, can contact APsystems or their merchant. However I think most merchants don't even know what an API is 😅

I'll close this for now 👍

closed 2025-02-03 by SonnenladenGmbH