SonnenladenGmbH/APsystems-EZ1-API #28: [Bug]: TypeError when retrieving device power status

ccMatrix, 2024-08-14:
### What happened?

When the data from the API is loaded it is directly converted to int without a check if the response was successful thereby leading to a TypeError on conversion.

### The python error

```shell
Logger: homeassistant.helpers.entity
Source: helpers/entity.py:942
First occurred: 14:51:50 (4 occurrences)
Last logged: 18:21:52

Update for switch.solar_inverter_status fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 942, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1300, in async_device_update
    await self.async_update()
  File "/usr/src/homeassistant/homeassistant/components/apsystems/switch.py", line 43, in async_update
    status = await self._api.get_device_power_status()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/APsystemsEZ1/__init__.py", line 280, in get_device_power_status
    return Status(int(response["data"]["status"])) if response else None
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 10: ''
```

### Your python version

3.12

### Your OS inlcuding your version

Home Assistant 2024.8.1 in Docker

mawoka-myblock, 2024-08-14:
Yea, that's true, but the TypeError only occurs when the inverter returns an error, so it's not a huge problem here. Feel free to create a PR

Bpolitycki, 2024-08-19:
Fixed in PR #29

closed 2024-08-19 by mawoka-myblock