SonnenladenGmbH/APsystems-EZ1-API #30: Restart of daily production

patman15, 2024-08-18:
Hi @mawoka-myblock!

Referring to https://github.com/SonnenladenGmbH/APsystems-EZ1-API-HomeAssistant/issues/34 I now made a patch for the HA integration. Nevertheless, after being done I thought, probably this should better be fixed already in the API, i.e. in this repo. Would you agree/accept a patch for the [get_ouput_data](https://github.com/SonnenladenGmbH/APsystems-EZ1-API/blob/1f69c5ce9c5f3b20a461898b31d0e822b0b133bf/APsystemsEZ1/__init__.py#L167)  function?
Only issue is that the API is then not stateless any more versus some strange code in the HA integration that does not seem to belong there.
Please let me know your thoughts then I can prepare a patch accordingly. I'm still on the old repo due to this missing functionality. Thanks!

mawoka-myblock, 2024-08-18:
Thanks for bringing that back up! The thing is that hardly any user seems to be facing this issue. I'd accept a PR when the debounce is/would be opt-in.

```python
def __init__([...], debounce_values: bool = False):
```

I'd accept and appreciate that! I really appreciate the caution and thoughts you put into this library!

mawoka-myblock, 2024-08-19:
I really don't know where it's coming from. My inverter handles it without a problem.

[image]

mawoka-myblock, 2024-08-19:
I honestly don't consider this to be required as the (Home assistant integration](https://github.com/home-assistant/core/blob/dev/homeassistant/components/apsystems/sensor.py) doesn't use it and I've never heard anyone complain about that problem. I honestly really think your inverter is faulty. And just setting it in the init of the home assistant integration to true isn't a huge problem at all. Please just create the PR here and only debounce the daily values when the option is set in the init function.

patman15, 2024-08-20:
> I really don't know where it's coming from. My inverter handles it without a problem.
> 
Not sure, can you deep zoom-in on 3rd of May, especially morning?

> I honestly don't consider this to be required as the Home assistant integration doesn't use it.
>
Sorry, I lost track, what do you mean by 'it' that is not used by HA?

In my opinion it is a principle _issue_ (not saying it is dramatic :wink:) with the inverter. The inverter turns of all communication (WiFi, BT) when there is not enough power from solar for some minutes **and** it resets daily energy value. Thus, whenever the solar power drops so the inverter turns off the issue occurs. Obviously, this is unlikely during day, especially summer times. No doubt.

I might see the issue more easily because I have very low power panels connected, but as described above it is an inherent topic due to the inverter functionality. Please correct me if I got something wrong.