SonnenladenGmbH/APsystems-EZ1-API #32: [Question]: does this library also work with APsystems EZ1D ?

Ocramius, 2024-08-23:
### What happened?

Hey there!

Sorry in advance if I'm poking you, as an integrator that happened to do this work out of their own good will.

## Context - feel free to skip

I'm currently trying to understand whether I should embark in buying APsystems modules, and the EZ1 is a bit undersized for what I need to do (2*600W panels).

The EZ1D seems more suitable for this use-case (1800W capacity).

## Question: is EZ1D compatible?

Given the above, is the EZ1D usable with this module?

If you don't know, who would we ask?

Thanks in advance!

### Your python version

irrelevant

### Your OS inlcuding your version

irrelevant

mawoka-myblock, 2024-08-23:
All of the EZ1 inverters are supported by this library though some docs are missing. Refer to the implementation in Home Assistant for reference.

closed 2024-08-23 by mawoka-myblock

Ocramius, 2024-08-23:
wow, that was fast: thanks @mawoka-myblock!