SonnenladenGmbH/APsystems-EZ1-API #34: Lifetime production counter reset to 0 when reaching 540,6 kWh

skrue, 2024-09-16:
Hi,
some days ago my inverter reached a lifetime production of 540,6 kWh on one of the strings and the counter was reset to 0. Four days later, the second string also reached 540,6 kWh and was also reset to 0.

Here's a screenshot showing the issue, I forward all my sensor data from Home Assistant to InfluxDB and visualize it using Grafana.
[image]

It looks the same in Home Assitant:
[image]

Any suggestions? I am still using the "old" custom component for HA integration, but to me it kind of feels like a problem with the inverter itself...

mawoka-myblock, 2024-09-16:
Seems like a problem with the inverter itself, so there's nothing I can do, so feel free to get this resolved with APsystems directly. Good luck with that and, if you should find anything out, I'd love to be updated on this issue. But as it's not a problem with this library, I'll close this issue! But still thanks for bringing that up.

closed 2024-09-16 by mawoka-myblock

xathon, 2025-06-05:
Did you ever hear back from APsystems on that?

mawoka-myblock, 2025-06-06:
Not yet, unfortunately, but I'll keep trying.

mawoka-myblock, 2025-06-06:
Have you reached out to the APsystems support yet as this is, as I've already said, an issue from APsystem's side and so far, I'm getting the same treatment as any other normal customer.