SonnenladenGmbH/APsystems-EZ1-API #36: [Bug]: ReturnOutputData.__init__() got an unexpected keyword argument 'originalData'

maoLz, 2024-10-08:
### What happened?

Hello, I am going to use Home Assistant to integrate apsystem ez1, but the code reports an error. I checked the local interface return contains an extra parameter: "originalData". In your code, "__init__" failed.
[image]

### The python error

```shell
2024-10-08 14:31:44.497 ERROR (MainThread) [homeassistant.components.apsystems] Unexpected error fetching APSystems Data data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 354, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/apsystems/coordinator.py", line 46, in _async_update_data
    output_data = await self.api.get_output_data()
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/APsystemsEZ1/__init__.py", line 189, in get_output_data
    return ReturnOutputData(**response["data"]) if response else None
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ReturnOutputData.__init__() got an unexpected keyword argument 'originalData'
```

### Your python version

python3

### Your OS inlcuding your version

Home ASSISTANT OS 13.1 (RPI 5)

maoLz, 2024-10-09:
When I write the constructor manually, everything works fine. Can I submit a PR?

[image]

mawoka-myblock, 2024-10-09:
That's not how it works, we'll have to investigate why your inverter returns that data, as other don't. Which model and firmware version are you using specifically?

maoLz, 2024-10-10:
Thank you for your prompt reply, here is the photo of my inverter information:
[image]

mawoka-myblock, 2024-10-10:
Very interesting! I'd appreciate a PR, but you have to make sure that the `originalData` is **optional** as my inverter doesn't return that. Just get started with the PR and feel free to ask for help there!

maoLz, 2024-10-11:
Thank you, I have submitted a PR.

PR #42 (2025-02-12): Address additional fields in ReturnData to be discarded instead
Closes #36
closed 2025-02-12 by mawoka-myblock