SonnenladenGmbH/APsystems-EZ1-API #39: Using Bluetooth to retrieve data from EZ1?

bluepuma77, 2024-10-25:
Got an EZ1 from Sonnenladen, it seems the smartphone app can fetch data with "direct connect" via Bluetooth. Is it possible to use the BT method instead of using WiFi/IP? Does the Bluetooth API need to "connect", does it need authentication, is it BT LE, is it just broadcasting data?

Would be interested in using an ESP32 with MicroPython/CircuitPython to fetch data from EZ1 and forward it via LoRaWan.

mawoka-myblock, 2024-10-25:
> Got an EZ1 from Sonnenladen, it seems the smartphone app can fetch data with "direct connect" via Bluetooth. Is it possible to use the BT method instead of using WiFi/IP? Does the Bluetooth API need [...]

Sounds like a good plan, but the problem is that the inverter doesn't transmit any bluetooth data after a successful wifi connection and as I don't want to change a running system, I don't have any option to test things out. I'll keep this issue open as maybe some people could help me in this regard.

As you seem technical and if you would like to assist me, you could get the "GATTBrowser" app by Renesas on your android phone and connect to your EZ1 prior to wifi setup, but after the normal setup in the app. Then, I'd love to receive a screenshot of all the attributes and, if possible, try "reading" every readable attribute. That'd be pretty much all the information I'd need for developing the bluetooth API.

ccvca, 2024-11-09:
I've had exactly same idea, as the wifi connection is not really working. (Need a Wifi Repeater, which likes to crash form time to time)

At least in my case, bluetooth is still active, event if the Wifi is connected and the local API is used (Using the APsystem App and direct connect). But I've never setup the cloud, so maybe bluetooth is gone after logging in with a cloud account. I'll try to get some data. A BLE-Scanner can at least read the name.

mawoka-myblock, 2024-11-09:
Yea, I've also tried the BLE scanner and it also could only read the name, not more unfortunately.

ccvca, 2024-11-09:
Yeah, but there are some more services and a possibility to write. Maybe some kind of "code" needs to be written first.... At least I can't find a clue, that there is any Bluetooth classic involved (at least Android does not show it to me).

mawoka-myblock, 2024-11-09:
Pretty much the same for me. But as you can read in my original reply, I'm not that interested in that and also don't have the skills for Bluetooth reverse engineering (yet).