SonnenladenGmbH/APsystems-EZ1-API #43: [Bug]: APsystems EZ1-M Incorrect values before the inverter switches off

Niklas9000, 2025-02-12:
### What happened?

Before the inverter switches off at the end of the day, implausible values are output, which leads to incorrect calculations. I noticed this in Homeassistant.

Example:

sensor.apsystems_e07000063823_today_production,0.39927,2025-01-27T15:56:08.554Z
sensor.apsystems_e07000063823_today_production,0.2063,2025-01-27T15:57:29.844Z
sensor.apsystems_e07000063823_today_production,0.20631,2025-01-27T15:58:21.579Z
sensor.apsystems_e07000063823_today_production,0.20632,2025-01-27T16:00:18.519Z
sensor.apsystems_e07000063823_today_production,0.20633,2025-01-27T16:02:02.555Z
sensor.apsystems_e07000063823_today_production,0.20634,2025-01-27T16:03:47.519Z
sensor.apsystems_e07000063823_today_production,0.20635,2025-01-27T16:05:31.560Z
sensor.apsystems_e07000063823_today_production,unavailable,2025-01-27T16:06:44.152Z

It is not possible that the value for today's production is suddenly lower.

The reason seems to be that P1 and P2 behave differently before switching off. Sometimes both are set to 0 shortly beforehand, sometimes only one.

This behavior is also visible on a friend's system.

[image]
[image]
[image]

### The python error

```shell

```

### Your python version

Home Assistant OS

### Your OS inlcuding your version

Home Assistant OS

retoch44, 2025-03-27:
I have the same problem (V1.7.0). I clicked through my energy archive and it started on December 6, 2024. Like you, the dropouts always occur shortly before shutdown. They occur 1 or more times a week. Since I always install new HA updates relatively quickly, could it have something to do with changes in HA? The firmware version has been identical here since summer 2024.

mawoka-myblock, 2025-03-28:
I can tell you guys for sure that it's the inverter's error. Nothing has changed in HA in this regard, so this is an APsystems issue. I'd first like to get this addressed by APsystems since it's their official API providing garbage data. If they don't care, then we can consider fixing their garbage data with our own filtering here in HA.

retoch44, 2025-03-30:
I bought my EZ1-M in May 2024 and since then it has been running in local mode without any problems until December. Why would the API work for 7 months without problems and then suddenly show this behavior?

xathon, 2025-04-08:
Just checked the HA repo, and it does actually get enabled there:
https://github.com/home-assistant/core/blame/12fc458abb85035d599222725149cd8678363b92/homeassistant/components/apsystems/__init__.py#L27

But I think this doesn't fix it for existing integrations, since there it wouldn't have been set during setup and default to `False`? I can't really check that and don't want to throw away my history data by removing and re-adding the device (or can I keep the history somehow?)

retoch44, 2025-04-09:
@xathon, thanks for your investigation. I reported this issue to APsystems support last week and they promised me to forward it to their development department. Since end of March I have had this error every single day and because the error accounts for about 20% of my actual day performance, I also correct the statistics for every single day. So I am the home assistant's assistant. :-)

[image]

I suspect that this could also be a hardware fault, e.g. ageing (inferior) components.

xathon, 2025-04-09:
@retoch44 how do you do the corrections?

But as mentioned, I really don't think that APsystems can actually do anything about that. I'll try to get some more detailed logs of what's going on, and why the debouncing doesn't trigger.

retoch44, 2025-04-09:
In Developer tools / Statistics, click on ‘Adjust total’ on the far right of the today_production sensor. A click on 'Outliers' will then show most of the errors that can be set to 0.

As I have no idea if my translation shows the original english menue items: Entwicklerwerkzeuge / Statistik, Summe anpassen, Ausreisser.

xathon, 2025-04-10:
Thanks! Lot of manual adjustment needed here.

Also, I noticed the same as you - this started on December 1st. And the commit that added the debouncing in the HA repo (https://github.com/home-assistant/core/commit/5deba1766e402ef316e4d98dd702b293a7d3423a) matches the timing, so this probably got introduced with 2024.12.0.

@patman15 could you have a look at this?

xathon, 2025-04-11:
I think the issue might be with these checks: https://github.com/SonnenladenGmbH/APsystems-EZ1-API/blob/9023ad09fadf32b17bee7a375db3bd8996527182/APsystemsEZ1/__init__.py#L132 https://github.com/SonnenladenGmbH/APsystems-EZ1-API/blob/9023ad09fadf32b17bee7a375db3bd8996527182/APsystemsEZ1/__init__.py#L144

What's probably happening here is something like this:
17:00: HA queries EZ1, reply is e1=2.0 kWh (e.g.)
17:01: EZ1 powers down
17:04: EZ1 powers back up, resets its counter
17:05: HA queries EZ1, reply is e1=0 kWh. Debounce would trigger, since state.old_state (2.0 kWh) > new_state (0 kWh). But **maybe this 0 value doesn't get interpreted as a float, but as an int**, causing the two `isinstance(new_state, float)` checks to fail (typed languages ftw). Thus the aggregated energy drops to 0 on the e1 panel, which also affects the summed energy stats and causes these spikes. Since HA doesn't expect an aggregated value to drop, it assumes that it must have been reset to 0 and records the unreasonably high statistic we see in the energy dashboard.

patman15, 2025-04-11:
> @patman15 could you have a look at this?

Will do. Noticed the same, but didn't have the time yet to check.

xathon, 2025-04-11:
> > @patman15 could you have a look at this?
> 
> Will do. Noticed the same, but didn't have the time yet to check.
> 
> EDIT: I think the issue is more difficult:
> [image]
> 
> Interestingly the inverter resets only one production, but keeps the other channel. That causes the issue. Need to think how to handle that correctly.

I'll check the raw API data tonight, but is it possible that, if only one panel gets sunlight, the API response doesn't actually include a value for e1/2? Though from looking at the code, seems like that would be caught by setting it to 0.0

patman15, 2025-04-11:
I think the debouncing is not catching that case and the effect comes actually from the reset of the counter from the inverter. So actually my code is failing to compensate. Reason could be really integer vs. float interpretation of `0`. All my tests use `0.0` (float) as value and not `0` (int). If the JSON parsing returns `int` that could be the reason for the debouncing to not work and then the energy dashboard sees the change and adds it up.

xathon, 2025-04-11:
Alright, I logged tonights API calls. This is what it looked like:
```
{"timestamp": "2025-04-11T20:33:19.042998", "status": "OK", "data": {"data": {"p1": 0, "e1": 1.76505, "te1": 175.02294, "p2": 0, "e2": 1.85579, "te2": 217.24654}, "message": "SUCCESS", "deviceId": "E17000041895"}}
{"timestamp": "2025-04-11T20:33:34.043063", "status": "OK", "data": {"data": {"p1": 0, "e1": 1.76505, "te1": 175.02294, "p2": 0, "e2": 1.85579, "te2": 217.24654}, "message": "SUCCESS", "deviceId": "E17000041895"}}
{"timestamp": "2025-04-11T20:33:49.043168", "status": "ERROR", "error": "HTTPConnectionPool(host='192.168.30.22', port=8050): Max retries exceeded with url: /getOutputData (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x7f398898cf90>, 'Connection to 192.168.30.22 timed out. (connect timeout=5)'))"}
{"timestamp": "2025-04-11T20:34:04.043310", "status": "ERROR", "error": "HTTPConnectionPool(host='192.168.30.22', port=8050): Max retries exceeded with url: /getOutputData (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x7f398898c610>, 'Connection to 192.168.30.22 timed out. (connect timeout=5)'))"}
{"timestamp": "2025-04-11T20:34:19.043433", "status": "OK", "data": {"data": {"p1": 0, "e1": 0.0, "te1": 175.0229, "p2": 0, "e2": 0, "te2": 217.2465}, "message": "SUCCESS", "deviceId": "E17000041895"}}
{"timestamp": "2025-04-11T20:34:34.043542", "status": "OK", "data": {"data": {"p1": 0, "e1": 0.0, "te1": 175.0229, "p2": 0, "e2": 0, "te2": 217.2465}, "message": "SUCCESS", "deviceId": "E17000041895"}}
{"timestamp": "2025-04-11T20:34:49.043701", "status": "OK", "data": {"data": {"p1": 0, "e1": 1e-05, "te1": 175.02291, "p2": 0, "e2": 0, "te2": 217.2465}, "message": "SUCCESS", "deviceId": "E17000041895"}}
{"timestamp": "2025-04-11T20:35:04.043776", "status": "OK", "data": {"data": {"p1": 0, "e1": 1e-05, "te1": 175.02291, "p2": 0, "e2": 0, "te2": 217.2465}, "message": "SUCCESS", "deviceId": "E17000041895"}}
{"timestamp": "2025-04-11T20:35:19.043845", "status": "OK", "data": {"data": {"p1": 0, "e1": 1e-05, "te1": 175.02291, "p2": 0, "e2": 0, "te2": 217.2465}, "message": "SUCCESS", "deviceId": "E17000041895"}}
{"timestamp": "2025-04-11T20:35:34.043929", "status": "OK", "data": {"data": {"p1": 0, "e1": 1e-05, "te1": 175.02291, "p2": 0, "e2": 0, "te2": 217.2465}, "message": "SUCCESS", "deviceId": "E17000041895"}}
{"timestamp": "2025-04-11T20:35:49.044005", "status": "OK", "data": {"data": {"p1": 0, "e1": 1e-05, "te1": 175.02291, "p2": 0, "e2": 0, "te2": 217.2465}, "message": "SUCCESS", "deviceId": "E17000041895"}}
{"timestamp": "2025-04-11T20:36:04.044115", "status": "ERROR", "error": "HTTPConnectionPool(host='192.168.30.22', port=8050): Read timed out. (read timeout=5)"}
{"timestamp": "2025-04-11T20:36:19.044187", "status": "ERROR", "error": "HTTPConnectionPool(host='192.168.30.22', port=8050): Max retries exceeded with url: /getOutputData (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x7f3988995210>, 'Connection to 192.168.30.22 timed out. (connect timeout=5)'))"}
{"timestamp": "2025-04-11T20:36:34.044306", "status": "ERROR", "error": "HTTPConnectionPool(host='192.168.30.22', port=8050): Max retries exceeded with url: /getOutputData (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x7f3988995590>, 'Connection to 192.168.30.22 timed out. (connect timeout=5)'))"}
{"timestamp": "2025-04-11T20:36:49.044390", "status": "ERROR", "error": "HTTPConnectionPool(host='192.168.30.22', port=8050): Max retries exceeded with url: /getOutputData (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x7f39889a1910>, 'Connection to 192.168.30.22 timed out. (connect timeout=5)'))"}
```
Note that e1 has 0.0 and e2 has 0. Funky. I'd guess that's because P1 got a smidge of light and P2 got nothing.

This is what Home Assistant made of it:

[image]
(at 20:35 the yellow / total line is at 1.77)

This confirms to me that the cast to float in #50 should fix this.