SonnenladenGmbH/APsystems-EZ1-API #49: Feature Request: Local access to API while using with Sunlit BK215 System.

de-sascha, 2025-04-02:
# Feature Request: Parallel Use of Local API Access and Sunlit BK215 System

## Description

Currently, it is not possible to use the local API access **and** have the inverter remain online and fully integrated in the Sunlit BK215 battery system (as shown in the Sunlit app) at the same time.

## Observed behavior
- When local API access is enabled, the inverter appears **offline** in the Sunlit app and is no longer associated with the BK215 system.
- When local API access is disabled again, the inverter returns to an **online** state and is fully visible in the app.

## Expected behavior
- The inverter should remain **online and connected to the BK215 system** in the Sunlit app, even when local API access is enabled.
- It should be possible to use the **local API for third-party integrations and monitoring**, while **retaining all features and data access in the Sunlit app**.

## Why this matters
- Local API access is crucial for smart home integration (e.g., Home Assistant), monitoring, and local data logging.
- At the same time, users want to keep using the Sunlit app for visualization and system control.
- Supporting both modes in parallel increases the system’s flexibility and value for advanced users.

## System Info
- Battery System: Sunlit BK215  
- Inverter behavior changes when local API is enabled  
- App Version: [Please fill in if known]

## Requested feature
Enable **parallel use** of local API access and full integration with the Sunlit BK215 system (including cloud/app functionality), without the inverter going offline in the app when local API is turned on.

mawoka-myblock, 2025-04-02:
Please reach out to the APsystems support for that.

closed 2025-04-02 by mawoka-myblock

SonnenladenGmbH, 2025-04-02:
A short comment:
Sunlit is currently working on implementing a local API that will also be able to talk to the EZ1-M on local mode.
The whole Sunlit System will eventually be available with full-local control using your own code or a HA Integration we'll write for it. This is planned within Q2 2025 along a lot of other software features.